eremitein/treble-patches #3: CAOS - No MTP (File Transfer)

Exetreezz, 2020-04-16:
Hi.

I am new to this kind of things (installing custom ROMs), but I really need to ask you about this issue, and since I couldn't find a "send message" option here, I will send this message in this way (please forgive me if this is not the place to do so).
The problem is this one and I really, really need an answer from you, I beg you!

After I downloaded and installed many custom ROMs, I found it yours and test it, and I think that yours is the cleanest android 10 ROM among them all (just 10 preinstalled necessary stock applications, no other "garbage" apps, and I congratulate you for that - I am so sick of tens Google and Samsung bootloader and stock garbage applications, which can get rid off them).

After I installed your latest "caos-v215-arm64ab-200414-vanilla" ROM on my "Samsung A40 (SM-A405FN)" phone, no matter what setting I apply in normal or Developer Options modes, I just can't make my phone to be visible on my laptop. All the USB ports are fine, my USB cable is good, the Samsung and ADB drivers are installed properly, everything is just fine on my side. In your ROM the "File Transfer" is set as preferred options (I tried the other ones as well, of course). I enabled or disabled the "USB Debugging" in "Developer options", and I even tried in unrooted and rooted mode too. Nothing is happening on my laptop screen: no internal memory and/or external SD card is visible on my desktop!
If I install other official ROMs or custom ones, I don't have this issue and my phone become visible on my desktop as soon I connect it with the same USB cable.

What am I missing here, please?
Is it a bug in your ROM or there is some "workaround" (a patch file or something) to be applied for the "File Transfer" (MTP) to be 100% functional, and my phone to be visible on my desktop, as it should normally be?

Please (I beg you) what is your advice regarding this issue?
Can you provide me a small "step by step" guide to fix that?
I really, really need it!

I thank you for your great work, with this the cleanest android ROM I've seen and tested so far!

P.S.: Do you know other bugs that my occur using this ROM further, please?

Have a great day! :)

eremitein, 2020-04-16:
hi, thanks!
for me MTP works well in CAOS.
as i know, Samsung phones have that problems with GSI roms. wait, maybe Phhusson will fix it soon. i can't help with that, i even haven't Samsung device

btw: there is a telegram group https://t.me/joinchat/CdHnpVThoZCgvPZx7ESNBA

Exetreezz, 2020-04-16:
Hi and thank you for reply be back.
I can't follow the Telegram for many days from now on, because I am preparing my phone from scratch, for using it fully.

So, you don't have a fix for that, too bad! :(

Do you know another very, very clean ASOP ROM (GSI preferably), just with a few stock apps, like yours, for testing it, please? Thanks in advance for that!

You know that you have other bug in your CAOS?
I installed "Viper4Android" (by flashing it with Magisk) and after I test it, I disabled it and deleted from Magisk, and after I restarted my phone (after the uninstall took place), the "Viper4Android" application was still installed in my android CAOS (not in Magisk anymore), and even if I tried many times to uninstall it, the app didn't uninstall no matter what, and it was "alive" all the time.

eremitein, 2020-04-17:
> Do you know another very, very clean ASOP ROM (GSI preferably), just with a few stock apps, like yours, for testing it, please?

i know only one another simple GSI rom, is a LineageOS

eremitein, 2020-04-17:
oh, and i guess you already know about Phh AOSP GSI

Exetreezz, 2020-04-17:
They're not as cleneast as yours!
I tried them both!

RealMahmoud, 2020-07-20:
i am having same issue with havoc gsi on samsung a30 😐

JuanTamadski, 2021-05-04:
If anyone here still have mtp not working, I suggest you to use custom kernel that supports mtp. Thats what I did to my Sam A505GN

closed 2021-05-27 by eremitein

GUELADIO67, 2022-02-17:
Hi. The files can be transfer in recovery mode. The device appear. You can try this solution and wait the boss PHHUSSON to find solutions for us. I have the same problem in Samsung A127F/DS(a12s). Android 11 GSI ROM after installation. It work nice the ROM is smooth clean and work finely with magisk root. Only USB missed seen in Windows. Thanks