eremitein/treble-patches #5: Request: Include Bromite Webview

JonnyTech, 2020-04-30:
Thank you for creating these images. Since you included the signature patch in CAOS (`Update (2020.02.27) - built with signature spoofing patch`), it is now my daily driver.

Can I request that you replace the built-in webview with Bromite (https://github.com/bromite/bromite) for those of us with privacy concerns that do not want anything Google on our devices?

Just replace `/system/product/app/webview/webview.apk` with the appropriate `xxx_SystemWebView.apk` from the Bromite repository. The end user should notice no difference at all.

JonnyTech, 2020-05-01:
> The end user should notice no difference at all.

Scratch that - it breaks SafetyNet :/

eremitein, 2020-05-01:
> > The end user should notice no difference at all.
> 
> Scratch that - it breaks SafetyNet :/

hi, why you think so? 
i tested today, all worked well

JonnyTech, 2020-05-01:
It states it in the wiki:

eremitein, 2020-05-01:
Oh, its' hacks way to install:)
and the information is old

> It states it in the wiki: https://github.com/bromite/bromite/wiki/Installing-SystemWebView#installation-options
> 
> But I do not have a need nor do I use safetynet so could not check.

closed 2020-05-01 by eremitein

JonnyTech, 2020-05-01:
Ah, so does that mean that you will include it? I have used it for ages on many phones and ROMs without issue. The less big-brother can "phone home", the better.

eremitein, 2020-05-02:
yes probably, in LOSQ already included