eremitein/treble-patches #7: Please add LiveDisplay and app-fullscreen display feature

hohommx2015, 2020-05-22:
This is the most less bug-free rom i've ever flashed,kudos to your awesome work,the only features missing from other roms  i wanted most are those features mentioned in the title:

1.LiveDisplay
2.App-fullscreen display

Also minimum brightness is too high for my eyes,can you adjust it to a lower value?

Please! Thanks in advance! I think ill stick with your rom forever!

JonnyTech, 2020-05-22:
CAOS/AOSP has `Night Light` which AFAIK is equivalent to `LiveDisplay`.

JonnyTech, 2020-05-22:
For display size, although not exactly what you asked, have you tried `Phh Treble Settings > Misc features > Compatibility Max Aspect Ratio > Infinite`

eremitein, 2020-05-22:
> Also minimum brightness is too high for my eyes,can you adjust it to a lower value?

hi, 
this is can be tuned in device overlay. see sample how to create/mod treble overlay https://github.com/phhusson/treble_experimentations/wiki/How-to-create-an-overlay%3F
for other questions JonnyTech gave you answer. by the way, what do you mean of "App-full screen display"?

hohommx2015, 2020-05-22:
> hi,
> this is can be tuned in device overlay. see sample how to create/mod treble overlay https://github.com/phhusson/treble_experimentations/wiki/How-to-create-an-overlay%3F
> for other questions [...]

Sorry my English is too bad to express the exact meaning,i mean the setting that let users force apps display in fullscreen.