eremitein/treble-patches #9: BLESS-GApps

drago06, 2020-06-04:
Hello. I have an Alcatel 1S (5024D), and there is no custom recovery for it. I can't place all the needed files in the system partition because in this ROM it's too small. I wanted to ask you if you could make a gapps version of the rom. Thank you very much in advance, you would help me a lot, I love your rom.

eremitein, 2020-06-06:
hi, i can't promise

drago06, 2020-06-06:
Hi, thanks for replying. I tried to include all the apks and frameworks in the unpacked system, but I can't find a way to pack it back. If you told me how to do just this I could test it and eventually send it to you so that you can share it with the other users

eremitein, 2020-06-06:
depends on how you unpack/modify it. there is a lot information on XDA, you can search there. and if you want you can join to my telegram group, link on the Treble GSI List page

philalat, 2020-06-17:
good morning 
i download your last BLESS rom but:
- i can't install gapps micro?
 - and i tried call BT but it's do no works.

philalat, 2020-06-20:
Hi
i love the boot moving image very sweet! i try to download another gapps= nano and i will see.
for call BT do you have an idea.
best regards

eremitein, 2020-06-20:
hello

1. before install gapps do automatic system resize in twrp
2. try workaround in Phh settings, search something like audio policy or bluetooth fix

philalat, 2020-06-20:
Hi
1 i do that in twrp but i always have error 70.
2  in phh setting/setting i open Misc feature /bluetooth workaround and i choice Mediatek.
whatelse :)

philalat, 2020-06-21:
good morning, afternoon or evening according to your location
i downloaded the last version of your rom (v220)
forgapps it's good with nano version, but for sound call BT always nothing. I have sound for music player or movies but nothing foe calls.
do you have an idea
best regards

eremitein, 2020-06-21:
hi
is a known problem with BT calls in all Q GSI, read project page
you can take logs and send there https://github.com/phhusson/treble_experimentations/issues

philalat, 2020-06-21:
Hi Victor
Thank you for your answer. I read about call BT problem, but in the
presentation of BLESS ROM it's written that call BT were fixed.
Good night

eremitein, 2020-06-21:
oh, this is just fix for regression in 2020.06.05 build.
i forget add generic bug section. project page will be updated, sorry

> but in the presentation of BLESS ROM it's written that call BT were fixed

philalat, 2020-06-22:
hi victor
could you give me your location to avoid waking up? for the BT problem
could you say me on the root stock system where i'll can see the folder
which have the BT?
because on my stock's rom the BT worked, so as a beginner i think if i can
compare the both, I'll find the missing files.
But you already thought about this.
best regards

eremitein, 2020-06-22:
i don't know exactly
you can look at system/etc/bluetooth
but i guess deal not in config, but in stock binary blobs

philalat, 2020-06-22:
Hi victor
could you say me what name of text editor i could use on my PC to open root system folder?
because i tried with note pad but the folder is too big for it.
thanks

eremitein, 2020-06-22:
i can't understand you. why you want open root system folder in text editor?..

philalat, 2020-06-22:
To see if it's possible the differences between original ROM and
yours.finally see what's the making folder.

philalat, 2020-06-22:
But I haven't your knowledge about this. I want to understand what's don't
work that all.I'm curious.

andrey167, 2020-06-29:
Hi, I have the same problem.my device can't have twrp support due to stock android 10, and the problem with gaps is very critical for me.I would like to have gapps already present in these builds

Cyberavater, 2020-08-08:
I'd appreciate it too if you could add a GAPPS version, Thanks for your work!

Cyberavater, 2020-08-08:
>before install gapps do automatic system resize in twrp

@eremitein By the way, How to do automatic system resize in twrp you mentioned?

closed 2021-05-27 by eremitein