eremitein/treble-patches #12: Fingerprint not working after reboot

Cyberavater, 2020-08-05:
Fingerprint works in the first boot but it doesn't in farther boots. Please tell me how to collect the debug log so that I can share it here.

Stock rom: https://cloud.mail.ru/public/46WH/m65NPMvAR/

eremitein, 2020-08-05:
no need logs, it's a known GSI issue (also exist in AOSP Quack) - some devices can't using fingerprint if used decrypted userdata. as i know, there is no bugfixes (i can't check this issue, because on my phone fp works even with decrypted data). but, there is one ugly workaround - you can using "SystemUI Restart" instead "Reboot phone"

Cyberavater, 2020-08-05:
But my phone is decrypted, tho. And I wasn't able to reproduce it on Descendant X, so....

eremitein, 2020-08-06:
> But my phone is decrypted

that's i said already - you should **crypt** userdata if you want using fp without problems

> And I wasn't able to reproduce it on Descendant X

show me logs and video, i need confirmation. because even AOSP GSI have this issue. anyway, Descandant is closed source ROM, so...

Cyberavater, 2020-08-06:
Okay, I have tested it 3 times on Descendant X and it works and doesn't even work on AOSP (I thought would work there since it worked on Descendant X and it was a CAOS only issue but I was wrong). Plz, tell me how can I share log, also, it'd be nice to ask him directly since it's a closed source rom.

eremitein, 2020-08-07:
i don't need your logs. you can ask developer of Descendant push his fix to the phhusson's AOSP github

Cyberavater, 2020-08-07:
I don't _think_ he will listen to me (not to be rude but he seems a little arrogant) but I've opened an issue in phh quoting him tho;
https://github.com/phhusson/treble_experimentations/issues/1432

Let's see if it helps.

closed 2020-08-07 by Cyberavater