eremitein/treble-patches #16: critical memory card problem

Maxym4uk, 2020-12-15:
Meizu m2 mini, MTK 6735, ARM64 A, VENDOR 9 ANDROID
Installed firmware - CAOS 11.0 v300.j
Sometimes, when you insert a memory card, it reports a problem (the memory card itself works fine) and offers to solve the problem.  But when I click to solve the problem, all the data is erased from the memory card.

eremitein, 2020-12-15:
Well, check in AOSP v300j

Maxym4uk, 2020-12-15:
> Ну, перевірте AOSP v300j

Well, on Aosp everything is fine

Maxym4uk, 2020-12-17:
> Що ж, перевірте AOSP v300j

So you will fix?

eremitein, 2020-12-17:
i haven't that issue, on my device sd-card works fine (vfat fs)
[image]

closed 2021-05-27 by eremitein