eremitein/treble-patches #17: ADB "unauthorized"

DaniV5, 2020-12-20:
can't get to use adb
the only thing i get is "Device name" unauthorized
using arm_avN build from 2020.10.29

Haus3master, 2021-06-17:
I also am getting the same error in lir mod, and since I can't access root and my device bootloops if force encrypt is disabled, so, yeah...

eremitein, 2021-06-17:
> can't access root

why? in any my Android 11 GSI root can be enabled in Phh settings > Misc > Superuser and then you can using any treminal app right from phone

also issue with ADB is not generic, in many devices it works like charm, so you need take a look on your device's vendor (seach in web howto force enable ADB debug)

Haus3master, 2021-06-17:
> why? in any my Android 11 GSI root can be enabled in Phh settings > Misc > Superuser and then you can using any treminal app right from phone
>
> also issue with ADB is not generic, in many devices it [...]

The superuser for some reason, didn't work with my device anyway, and for the adb issue, I tried to go to a newer vendor image, but, due to a cable disconnect due to setting the phone down caused the upgrade to fail and now my preloaded is gone and the phone is dead, I tried to look for test points on the internet for the Nokia 1 (My device), but I didn't find any, and I don't want to totally kill it by shorting random pins and hoping for it to work without a multimeter for measurements, and if the test ports aren't on the surface but rather than under the shielding, it's basically dead since pulling the shielding out makes it unusable, (edl cables don't work since my device is mtk based), so I can't do anything at all, so, consider my comment irrelevant, sorry about this...