eremitein/treble-patches #18: Request: Include SeedVault to crDRom project (and others)

kikecalpe, 2021-01-04:
Thanks for your work.
Can you include Seedvault to your builds?
It must be build with AOSP, [here's the guide](https://github.com/seedvault-app/seedvault/wiki/Building-with-AOSP).

eremitein, 2021-01-04:
you sure it must be build with AOSP?.. first try pre-built apk from https://github.com/seedvault-app/seedvault-prebuilt (use adb shell commands from guide, you need root and put Seedvault app inside system priv-app)

kikecalpe, 2021-01-04:
Well, [Readme](https://github.com/seedvault-app/seedvault#what-makes-this-different) says that, but I've made an assumption, that is for no rooted systems.
I've tried as you say; rooted with magisk 21 phh, created 'Seedvault' folder in priv-app (with 755), copied files in there (with 644), applied both bmgr commands, and reboot.
Now backup shows under system settings as active, but when I open it there's a black screen.
Anyway, thanks for your help.