eremitein/treble-patches #22: [Honor v20]  The ringtone is still ringing during the call.

ijustyce, 2021-01-28:
First of all, thanks  for your excellent work, but I ran into a problem. After I answer the call, the ringtone of the call is still ringing. The volume of the ringtone became smaller, but it did not disappear. I use Honor v20 with latest BLESS rom.

I am new to gsi, can you help me fix it, thanks in advance.

eremitein, 2021-01-28:
Hi, ask in [telegram support group](https://t.me/joinchat/CdHnpVThoZCgvPZx7ESNBA), there is some users have Honor and maybe they are can help. I haven't that issue on my device, so no idea why this happens

ijustyce, 2021-02-02:
> Hi, ask in [telegram support group](https://t.me/joinchat/CdHnpVThoZCgvPZx7ESNBA), there is some users have Honor and maybe they are can help. I haven't that issue on my device, so no idea why this happens

Ok thanks. By the way, how to build bless rom based on your code. I can't find the branch or tag. Thanks again.

eremitein, 2021-02-02:
it's merged in project https://gitlab.com/android-generic/

ijustyce, 2021-02-07:
Finally I fixed it by use China Mobile (The CMCC) sim card. 
I remember there are GsmPhone.java and CdmaPhone.java in the past so I insert the CMCC sim card into my phone just for test, and it work fine. Don't know why China Telecom has this strange issue. 

Anyway, thanks for your great work. I built it from your code and ran it successfully. May be I will fix the CDMA sim card issue someday, if all done I will make a merge request. Thanks again.

closed 2021-02-07 by ijustyce