eremitein/treble-patches #30: Screen rotation is not working properly

acastillorobles77, 2021-03-13:
Hi, there is the problem with my tablet: screen rotation is not working properly, this is how it behaves:

Portrait = landscape
Landscape reverse = portrait
Landscape = portrait reverse
Portrait reverse = landscape reverse

This only happens on Android 11 GSIs, no problem with Android 10 GSIs for example.

Everything except rotation seems working fine. How I can fix this?

eremitein, 2021-03-13:
hi, first check it in AOSP v302 by phh
if issue still persist - send bugreport to phh

acastillorobles77, 2021-03-16:
Alright, so how I can make a bugreport and how to send it to phh? Sorry I'm a noob :p

eremitein, 2021-03-17:
> Alright, so how I can make a bugreport and how to send it to phh? Sorry I'm a noob :p

https://github.com/phhusson/treble_experimentations/issues
also search in buglist, maybe already reported