eremitein/treble-patches #32: `Invalid sparse file format at header magic` Error

banx, 2021-04-02:
When I try to flash the `lir-v302-210318-arm-avZ.img` Image to the system partition, I get the error mentioned in the title.
When the flashing is complete, the device goes into a boot loop.
Is it a problem with my device or is it a problem in the way the Image file was built?

Note, flashing LOS 17.1 didn't have this problem.

JonnyTech, 2021-04-02:
What device?

banx, 2021-04-02:
Lenovo E10

banx, 2021-04-02:
Can you help me understand what's the difference between `aoZ`, `agZ` and `avZ` images?

JonnyTech, 2021-04-02:
From the Telegram group (https://t.me/joinchat/CdHnpVThoZCgvPZx7ESNBA):

```
Information about what type of GSI it is

<ARCH>_xyZ{-lite}

<ARCH> can either be arm, a64 or arm64
x can either be a or b
y can either be v, o, g or f
Z can be N, S or Z

b = a/b
a = a-only

g = gapps
o = gapps-go
v = vanilla (no gapps included)
f = floss (free & open source apps instead gapps)

N = no superuser
S = superuser included
Z = enhanced dynamic superuzer

-lite = Android11 for vndklite vendors

so in most common example "arm64_bvN" is ARM64 A/B vanilla with no superuser
```

banx, 2021-04-02:
Is [sas-creator](https://github.com/phhusson/sas-creator) used to generate the A-only images here?
I couldn't help noticing the final line in [Andy's repo README](https://github.com/AndyCGYan/treble_build_los), which make me wonder if this could be the cause. I've also tried Andy's images, and they generated the same error.

JonnyTech, 2021-04-02:
Yes, system-as-system creator is used in all Android 11 builds: https://github.com/phhusson/treble_experimentations/blob/master/build.sh

eremitein, 2021-04-02:
> When I try to flash the `lir-v302-210318-arm-avZ.img` Image to the system partition, I get the error mentioned in the title.

this is just notice, not error. anyway you can safely ignore it

> When the flashing is complete, the device goes into a boot loop
> Note, flashing LOS 17.1 didn't have this problem

first check with AOSP by phhusson, if same problem - send him bugreport
(but first you need to be sure there is no Magisk in boot partition)

banx, 2021-04-03:
The same problem persists with AOSP by @phhusson. I'll raise this issue there.

> (but first you need to be sure there is no Magisk in boot partition)

No Magisk, the original boot partition is used.

Nothing-Dev, 2021-04-14:
That warning is normal, your bootlop is caused by another thing. I recently flashed LiR throught fastboot, get the same warning and nothing happened, the room boots normal.

banx, 2021-04-17:
> That warning is normal, your bootlop is caused by another thing. I recently flashed LiR throught fastboot, get the same warning and nothing happened, the GSI booted without issue.

@Nothing-Dev Do you have any suggestion that may help investigating the cause of the bootloop?

Nothing-Dev, 2021-04-18:
Did you try to use permissiver file?

eremitein, 2021-04-18:
> > That warning is normal, your bootlop is caused by another thing. I recently flashed LiR throught fastboot, get the same warning and nothing happened, the GSI booted without issue.
> 
> @Nothing-Dev Do you have any suggestion that may help investigating the cause of the bootloop?

logs may help. also you sure you need A-only? check with [Treble Info](https://play.google.com/store/apps/details?id=tk.hack5.treblecheck) app

closed 2021-05-27 by eremitein

Bbolaji33, 2021-10-19:
How can I tell if magisk is in boot

Bbolaji33, 2021-10-19:
Cuz I tried lineage 7 on a tecno pouvoir 2 pro...it got stuck in boot loop