eremitein/treble-patches #38: libaptX is NOT open source

AndroPlus-org, 2021-09-16:
A reverse-engineered library should not be called open source.
The original developer also says "compilation and redistribution in a binary format is forbidden!".
Please pay respect to the original developer.
https://github.com/Arkq/openaptx

eremitein, 2021-09-17:
> A reverse-engineered library should not be called open source.
> The original developer also says "compilation and redistribution in a binary format is forbidden!".
> Please pay respect to the original developer.
> https://github.com/Arkq/openaptx

this is not my busy, in my builds used https://github.com/notmyst33d/libaptX and author called it "Open source replacement for proprietary Qualcomm aptX libraries on Android"

AndroPlus-org, 2021-09-18:
In "Credits” section, there is the link to the original project.
notmyst33d only did setup for Android building and libaptx code is Arkq's.

eremitein, 2021-09-18:
> In "Credits” section, there is the link to the original project.
> notmyst33d only did setup for Android building and libaptx code is Arkq's

again - not my busy, because notmyst33d pushed all project under MIT permissive license (and all files with code have proper copyright by Arkadiusz Bokowy, which code also under MIT), so i can using it in my GSI roms