eremitein/treble-patches #41: [Samsung Galaxy A22 - SM-A225M] Some issues

itaranto, 2021-12-11:
Hello everyone,

I'm using the crDRom11 from the treble patches, more specifically: crdrom-v313+211013-arm64-bvZ.img.

My device is a Samsung Galaxy A22 (SM-A225M 4G version), and I'm experiencing the following issues:

- The proximity/light sensor doesn't work, for example when listening an audio on Telegram
- The jack audio port doesn't seem to be working either.
- The mobile internet seems a little bit slow.

How can debug/troubleshoot any of this?

You thing another ROM may be better for this phone?

Thanks in advance.

eremitein, 2021-12-24:
> * The proximity/light sensor doesn't work

if magisk installed - remove magisk

itaranto, 2021-12-24:
> > * The proximity/light sensor doesn't work
> 
> if magisk installed - remove magisk

I haven't installed magisk.

I also checked under _Phh Treble Settings -> Misc features -> Debug sensors_, should I see the proximity sensor there?

eremitein, 2021-12-25:
> should I see the proximity sensor

i don't know. you have same issue in the AOSP v313 or latest LineageOS by Andy? check it first

itaranto, 2021-12-25:
> > should I see the proximity sensor
> 
> i don't know. you have same issue in the AOSP v313 or latest LineageOS by Andy? check it first

Mmm I only tried [CAOS11](https://github.com/eremitein/treble-patches/wiki/CAOS11-Project) and [crDRom11](https://github.com/eremitein/treble-patches/wiki/crDRom11-Project).

I may try those, I'll do it when I have time, since this phone is my daily driver.

eremitein, 2021-12-26:
links to Andy's GSI builds you can find on XDA forum