eremitein/treble-patches #45: Request: Add Crdroid settings / mods

nguadien, 2022-03-10:
Crdroid is inferior to this rom in terms of performance, however it has some nice features:

Status Bar & Quick Settings
Customize what icons you see as well as effects and number of QS rows and columns

Navigation and buttons
Choose from gesture nav or old school nav buttons as well as customize what your hardware buttons do

I especially love the "Hold Back to Screen off" and "Hold Recents to Torch" newly added in MIUI 12.5.4. Crdroid also has these features...

So I want to request, can you add the above Crdroid settings / mods to CAOS11?

Relected, 2022-03-10:
Also some havoc is customization features are also great, such as removing the x from the signal bar and customizing nav and status bar

nguadien, 2022-03-10:
> Also some havoc os customization features are also great, such as removing the x from the signal bar and customizing nav and status bar

Yes, it's also great to have in CAOS11.
CAOS11 now is the best custom ROM I got. Been using it daily and it works so bug-free and fluently. If it has customizations from Havoc + CrDroid, it would be in godlike tier, man.

eremitein, 2022-03-11:
> number of QS rows and columns

already have this, in Display menu you can select prepared sheets  
adding features is not so easy. currently CAOS already have more features than PE+ i guess